sandeepmistry/arduino-BLEPeripheral #262: Is there a way to restart ble work after BLEPeripheral.end()?

patrvanh, 2019-04-12:
Hi, 
I wish to save power in my implementation by turning the radio off for some periods when it is not used. I tried using the `end()` function to shut down all radio work and then again calling `BLEPeripheral::begin()` but it seems that the radio is not restarting. At least I'm not able to connect to the device.

zenpa, 2019-05-06:
Hi, me too, I have the same issue, after end() the device cannot restart radio communication after begin().
Any suggestion? Thank you for a feedback.

kozle, 2019-06-07:
I'm having the same problem. 
Did you guys found any solution?

YusukeKishimoto, 2020-05-04:
I was able to solve it by running `From git` and `Flashing a Soft Device` from the URL below.

https://github.com/sandeepmistry/arduino-nRF5

Also, at that time, since the file described in **softdevices.txt** does not exist, it was necessary to make the following changes.
```
s110.url=https://www.nordicsemi.com/-/media/Software-and-other-downloads/SoftDevices/S110/s110nrf51800.zip

s130.url=https://www.nordicsemi.com/-/media/Software-and-other-downloads/SoftDevices/S130/s130nrf51201.zip
...
```