sandeepmistry/arduino-BLEPeripheral #265: bytes after a 0x00 are not sent

zeroby0, 2019-06-26:
For value for `BLECharacteristic`, bytes after a `0x00` byte are not sent. This is probably because `BLECharacteristic` uses `char` for payload and `0x00` is being treated as the `'\0'` character.

What it means is that custom binary data cannot be sent. For example, `0x00 0xF0 0x00` is received as an empty message.

Here is an example sketch. You can check notifications via Web-Bluetooth at this url [https://googlechrome.github.io/samples/web-bluetooth/notifications-async-await.html?service=0xfff0&characteristic=0xfff1](https://googlechrome.github.io/samples/web-bluetooth/notifications-async-await.html?service=0xfff0&characteristic=0xfff1).

```
#include <SPI.h>
#include <BLEPeripheral.h>

char data[3];
char buffer[20];

BLEPeripheral blePeripheral = BLEPeripheral();
BLEService s_primary = BLEService("fff0");
BLECharacteristic c_data = BLECharacteristic("fff1", BLERead | BLEIndicate, 3);

void setup() {
  Serial.begin(115200);

  blePeripheral.setLocalName("Nrf52");
  blePeripheral.setAdvertisedServiceUuid(s_primary.uuid());

  blePeripheral.addAttribute(s_primary);
  blePeripheral.addAttribute(c_data);

  blePeripheral.begin();

  data[0] = 0x00;
  data[1] = 0xF0;
  data[2] = 0x00;
}

void loop() {
  BLECentral central = blePeripheral.central();

  if (central) {
    Serial.print(F("Connected to central: "));
    Serial.println(central.address());

    while (central.connected()) {
        if(c_data.setValue(data)) {
            sprintf(buffer, "data: %02X %02X %02X", data[0], data[1], data[2]);
            Serial.println(buffer);
        }

        delay(1000);
    }

    Serial.print(F("Disconnected from central: "));
    Serial.println(central.address());
  }
}
```

The same output as Web-Bluetooth is received in Nordic nRF Connect app as well.

Using `BLEFixedLengthCharacteristic` instead sends wrong data. For example, for `0x00 0xF0 0x00` as payload, I receive `0x32 0x99 0x1B`.

## Other details

I'm using the `nrf51-indicate-fix` branch. 

My chip is NRF52832, the issue with fixed length might be related to the different chip, I don't have an NRF51 chip to verify.

I'm using platformIO ide on vscode. Here is my platformio.ini file
```
[env:nrf52_dk]
platform = nordicnrf52
board = nrf52_dk
framework = arduino
monitor_speed = 115200
debug_tool = jlink
; SoftDevice version
build_flags = -DNRF52_S132
lib_deps =
  https://github.com/sandeepmistry/arduino-BLEPeripheral/archive/nrf51-indicate-fix.zip
```
With FixedLength size of 20, here is the output I get: `0x32 0x99 0x1b 0xc3 0xfe 0x9f 0xed 0xfc 0x24 0x00 0x00 0x00 0xfc 0xff 0xff 0xff 0x34 0x23 0x00 0x20`. This doesn't change even when payload changes.

atc1441, 2020-02-12:
Hi, you can make it work by giving a length with the setValue

Here as example: 

c_data.setValue(data,len);

zeroby0, 2020-02-13:
I see, thanks 😄

closed 2020-02-13 by zeroby0