sandeepmistry/arduino-BLEPeripheral #269: blePeripheral.begin()  not executing.

JMAC4790, 2019-07-25:
I'm using an nrf8001 module from adafruit to send pressure readings to a device using ble. I'm having an issue getting the code to execute the blePeripheral.begin(); line. I've included the code and the warnings/errors im receiving when verifying the code. Any help in rectifying this issue is appreciated as I can't work it out. 

Code

`#include <BLEPeripheral.h>
#include<SPI.h>
#include <Wire.h>

#define BLE_REQ 10
#define BLE_RDY 2
#define BLE_RST 9

BLEPeripheral blePeripheral = BLEPeripheral(BLE_REQ,BLE_RDY,BLE_RST);
BLEService service = BLEService("CCC0");

BLECharacteristic characteristic = BLEFixedLengthCharacteristic("CCC1", BLERead | BLENotify,12);

unsigned long previousMillis = millis();    //Stores the last time sensor was read
unsigned long interval = 1000;         //Interval between sensor readings

void setup() 
{
  Serial.begin(9600);
  Serial.println(service.uuid());
  Serial.println(characteristic.uuid());
  Serial.println(F("Bluetooth Low Energy Sensor Reader"));

  blePeripheral.setLocalName("Pressure Reader");  
  blePeripheral.setDeviceName("Pressure Reader");

  Serial.println(F("Local name and device name set to Pressure Reader"));
  
  blePeripheral.setAdvertisedServiceUuid(service.uuid());
  blePeripheral.addAttribute(service);
  
  blePeripheral.addAttribute(characteristic);

  Serial.println("1234");
  void clearData();
  blePeripheral.begin();
  Serial.println("blePeripheral.begin() worked");
}
void loop() 
{
  blePeripheral.poll();

  if (millis() - previousMillis >= interval)
  {
    pollSensors();
    previousMillis = millis();
  }
}
void pollSensors()
{
  union {
    float f;
    byte bytes[4];
  } floatUnion;

  union {
    float f;
    byte bytes[4];
  } floatUnion2;

  union {
    float f;
    byte bytes[4];
  } floatUnion3;

  
  float voltage1 = analogRead(A0);
  float pressure1 = voltage1*5/1023;      //Change for correct calibration

  floatUnion.f = pressure1;

  float voltage2 = analogRead(A0);
  float pressure2 = voltage2*2*5/1023;    //Change for correct calibration

  floatUnion2.f = pressure2;

  float voltage3 = analogRead(A0);
  float pressure3 = voltage3*3*5/1023;    //Change for correct calibration

  floatUnion3.f = pressure3;

  char byteArr[12];

  int ii = 0;
  for(ii= 0 ; ii < 12 ; ii ++ ){
    if(ii<4){
      //pressure1
      byteArr[ii] = floatUnion.bytes[ii];
    }else if((ii>=4)&&(ii<8)){
      //pressure2
      byteArr[ii] = floatUnion2.bytes[ii-4];
    }else if((ii>=8)&&(ii<12)){
      //pressure3
      byteArr[ii] = floatUnion3.bytes[ii-8];
    }
  }
 characteristic.setValue(byteArr);
    Serial.print(F("Pressure 1: "));
    Serial.println(pressure1);
    Serial.println(pressure2);
    Serial.println(pressure3);
    
}`

Warning/Error messages

`C:\Users\joshm\Documents\Arduino\libraries\arduino_958432\src/utility/hal_aci_tl.h:86:3: note: type 'struct aci_pins_t' should match type 'struct aci_pins_t' that itself violate one definition rule

 } aci_pins_t;

   ^

C:\Users\joshm\Documents\Arduino\libraries\arduino_958432\src/utility/hal_aci_tl.h:86:3: note: the incompatible type is defined here

 } aci_pins_t;

   ^

C:\Users\joshm\Documents\Arduino\libraries\arduino_958432\src/utility/lib_aci.h:65:16: note: type 'struct aci_state_t' should match type 'struct aci_state_t' that itself violate one definition rule

 typedef struct aci_state_t

                ^

C:\Users\joshm\Documents\Arduino\libraries\arduino_958432\src/utility/lib_aci.h:65:16: note: the incompatible type is defined here

 typedef struct aci_state_t

                ^`

John-Almardeny, 2020-05-14:
Here is the same. It looks the developer of this library has no clue !

jLynx, 2021-05-11:
I currently have the same issue on a generic NRF51822