sandeepmistry/arduino-BLEPeripheral #287: Why is my BLE Service not appearing after a successful flash?

ab3lson, 2021-03-13:
I am trying to flash a generic [nRF52823](https://www.aliexpress.com/item/32819293925.html?spm=a2g0s.9042311.0.0.69c94c4dxwezN9) with the [test](https://github.com/sandeepmistry/arduino-BLEPeripheral/blob/master/examples/test/test.ino) sketch. When I flash the soft device, it says that the programming finished and the verification was successful. Here is the log:

```
Open On-Chip Debugger 0.10.0-dev-00254-g696fc0a (2016-04-10-10:13)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
debug_level: 0
0x4000
adapter speed: 10000 kHz
nrf52.cpu: target state: halted
target halted due to debug-request, current mode: Thread 
xPSR: 0x61000000 pc: 0x0001b08e msp: 0x20001188
nrf52.cpu: target state: halted
target halted due to debug-request, current mode: Thread 
xPSR: 0x01000000 pc: 0xfffffffe msp: 0xfffffffc
** Programming Started **
auto erase enabled
nrf52.cpu: target state: halted
target halted due to breakpoint, current mode: Thread 
xPSR: 0x61000000 pc: 0x2000001e msp: 0xfffffffc
wrote 114688 bytes from file C:\Users\Admin\Documents\ArduinoData\packages\sandeepmistry\hardware\nRF5\0.7.0/cores/nRF5/SDK/components/softdevice/s132/hex/s132_nrf52_2.0.1_softdevice.hex in 2.602834s (43.030 KiB/s)
** Programming Finished **
** Verify Started **
nrf52.cpu: target state: halted
target halted due to breakpoint, current mode: Thread 
xPSR: 0x61000000 pc: 0x2000002e msp: 0xfffffffc
nrf52.cpu: target state: halted
target halted due to breakpoint, current mode: Thread 
xPSR: 0x61000000 pc: 0x2000002e msp: 0xfffffffc
verified 110636 bytes in 0.377083s (286.523 KiB/s)
** Verified OK **
** Resetting Target **
shutdown command invoked
```
That output leads me to believe that the sketch was uploaded successfully and the chip should start broadcasting over BLE. However, when I use a BLE scanning app, I don't see any device called "test". Is there anything else I need to do in order for the chip to start broadcasting?

adamhun94, 2021-03-22:
Previously I was able to flash my chip (exactly the same as yours), but through my Pi.
I think this is the correct config that I used, maybe you can use it:
```
source [find interface/sysfsgpio-raspberrypi.cfg]
transport select swd
 
set CHIPNAME nrf52
source [find target/nrf52.cfg]

reset_config  srst_nogate
 
adapter_nsrst_delay 100
adapter_nsrst_assert_width 100
 
init
reset halt
nrf5 mass_erase
reset halt
program /home/pi/nrf52/s132_nrf52_7.0.1_softdevice.hex verify
program /home/pi/nrf52/ble_app_hrs_c_pca10040_s132.hex verify
reset run
exit
```

It might contains unnecessary commands, but I was happy that it works 😄 
Note: I compiled an official nrf example with SES ide, not using this library.

Alternatively you can use the 'mergehex' tool which merges the two hex files and you only need to flash one file.

Note 2: Meanwhile I bought a J-link debugger, so I use it for programming. After flashing, I need to send two commands: **r**(=reset) and **g**(=run), which are I think identical to **'reset run'** in OpenOCD. But if you can't see you device, you might need to manually reset it (cut and re-add power).

ab3lson, 2021-03-24:
Thanks for all that I do. I may give that a go. The thing that stumps me is that the programmers output makes it look like the programming was successful. The test sketch should just broadcast a new service, but I don’t see it after I flash it. Do I need to send some command to wake it back up?

Also, when you flashed yours, did you just solder to the DIO, CLK, VIN, and GND, update the program and then search for the service? I am wondering if I should try to attach an FTDI adapter to check for serial communication to ensure that the flash is actually working.