sandeepmistry/arduino-BLEPeripheral #291: Freeze on blePeripheral.begin();

jLynx, 2021-05-11:
Hi 

I have just designed a NRF51822 board and all my test code seems to be working apart from getting bluetooth to work.

My code always seems to freeze on the line `blePeripheral.begin();`

Looking at the LED sketch, it freezes on line 38 https://github.com/sandeepmistry/arduino-BLEPeripheral/blob/161a4163f565be3cd5b62bbc59f0c2b522d82b02/examples/led/led.ino#L38

Here is my config:
[image]

(Note: Right now there is no antenna on the board as I am waiting for it to show up, but I doubt that will have anything to do with the code not executing) 

If you need any more information please let me know.

jLynx, 2021-05-11:
Seems like there are a few other people with the same issue, but lacking detail.

- https://github.com/sandeepmistry/arduino-BLEPeripheral/issues/269
- https://github.com/sandeepmistry/arduino-BLEPeripheral/issues/268
- https://github.com/sandeepmistry/arduino-BLEPeripheral/issues/257
- https://github.com/sandeepmistry/arduino-BLEPeripheral/issues/22 (Answered but no fix found)
- https://github.com/sandeepmistry/arduino-BLEPeripheral/issues/146

jLynx, 2021-05-11:
If I understand correctly, they should all be -1? I have tried this but still no success

jLynx, 2021-05-17:
As much as this has fixed the initial issue to get me going, I have found that the RC Oscillator is using the internal RC oscillator and not the External 32.768 kHz crystal (Crystal Oscillator). Will be doing more research into why its not able to start with this selected. I am assuming it something to do with this line
[image]
maybe with the accuracy being off with my crystal

LiveRock, 2022-05-31:
can we reconfig BLE_REQ, BLE_RDY, BLE_RST pins?