sandeepmistry/arduino-BLEPeripheral #294: RFDuino using BLEPerfipheralObserver to scan for iBeacons - Issue

fkahhaleh, 2022-01-18:
Hi,
Thank you everyone for your efforts in making this library what it is today.

I am reviving an old RFduino that I have instead of collecting dust for a smart home project.
Basically I want to have the RFduino run as an observer and scan for iBeacons around.

Running the BLEPeripheralObserver fork by floe, I got the following output from the observer.ino example:
`⸮Got own addr: c7:4a:8f:07:2d:bc
start scanning
`

The own addr:... shows up as soon as the  code is run, and when I send any characters (I just hit enter) through serial I get
the start scanning, however it is a fast executed one along side the call for .poll().
Basically if I keep hitting ENTER or any character I get "start scanning" so it really doesn't seem to be busy with scanning!

3 iBeacons are around me and none are being picked up.

I tried to remove the 'transmit' method calls to see if that has any affect and it didn't change anything.

Any feedback or suggestions on how to proceed?

Thank you
F.K.

floe, 2022-01-18:
Hi, author of the fork in question (https://github.com/floe/BLEPeripheralObserver) here - I've never used an RFduino, so I can't really suggest much. In theory, once scanning has started and a beacon is detected, it should trigger the callback here: https://github.com/floe/BLEPeripheralObserver/blob/master/examples/observer/observer.ino#L41-L49

fkahhaleh, 2022-01-18:
Hi floe,

Thanks for the feedback.

That's my understanding, however you made me think about HW compatibility with your comment about RFduino.
Therefore I dug a little deeper, and the nrf51822 class has the following:

```
void nRF51822::startScanning() {
#ifdef NRF_51822_DEBUG
  Serial.println(F("Start scanning"));
#endif

#ifndef __RFduino__
  // see https://infocenter.nordicsemi.com/index.jsp?topic=%2Fcom.nordic.infocenter.s130.api.v2.0.1%2Fstructble__gap__scan__params__t.html
  ble_gap_scan_params_t scanParameters;

  memset(&scanParameters, 0x00, sizeof(scanParameters));

  scanParameters.active      = 1;     // send scan requests
  scanParameters.interval    = 0x140; // 200 ms in units of 0.625 ms
  scanParameters.p_whitelist = NULL;  // no whitelist
  scanParameters.selective   = 0;
  scanParameters.timeout     = 10;    // 10 seconds timeout
  scanParameters.window      = 0xA0;  // 100 ms

  sd_ble_gap_scan_start(&scanParameters);
#endif
}
```

it completely bypasses the startScanning request if __RFduino__ is defined !!
So not sure what to make of it, is RFduino not supporting this since from what I read it's running a tad bit of an older softdevice or not.

I'll try to test this and see how it goes but in the meantime if anyone has any feedback on RFduino and the observer role please do share.
Thanks!